dhylands/rshell #147: (some) files cause rshell to hang when copying to a Pico

lornajane, 2021-02-08:
Using rshell (0.0.28 from pip) to access the Raspberry Pi Pico: `rshell -p /dev/ttyACM0 --buffer-size=512`

When using `cp` or `rsync`, some files cause rshell to hang, but others work fine. I am not sure what causes this, sometimes changing my python code helps, increasing the buffer size seemed to help too. I saw some similar issues open already around this (#144) and can confirm that the `pico` branch is working much better for me.

avian2, 2021-02-09:
I can confirm this happens with my Raspberry Pi Pico as well.

Doesn't seem to be related to binary files, since it hangs on some pure ASCII files as well. Specifying `-a` doesn't help.

Version installed from pypi using `pip install rshell` (as instructed by Pico manual) has this problem (0.0.28). Switching to `pico` branch from git indeed helps.

```
MicroPython v1.14 on 2021-02-05; Raspberry Pi Pico with RP2040
```

gavinB-orange, 2021-02-16:
I've hit this as well. Same rshell version / Pico.

A workaround for me is to increase the buffer size. Files which hang will start working if e.g. I increase buffer size to 2048

gavinB-orange, 2021-02-16:
Note - it's not as simple as file size, as I've tried copying large e.g. 3K binary data files, and I've never had a problem.  Python code files though - presumably as they are ascii text - are the issue.  Small files are fine. ~2.7K seem to break.

toybuilder, 2021-05-13:
I seem to have a similar issue with a ESP32 target -- I tried to use rshell as a way to bootstrap a batch of new boards with the same code base and found that rshell rsync would hang at some point.  If I reset and reconnect and do the rsync again, the same file that it hung on before would sync and rsync would continue a bit before choking again.

davehylands, 2021-05-13:
What's usually happening in this case is that a character gets dropped in the serial buffers and then rshell looses sync (it doesn't have any timeout/retry logic). Sometimes tweaking the buffer size can help. The ESP series is especially bad at this due to the OS running in the background sometimes disabling interrupts for long enough to allow serial characters to be dropped.

arbego, 2022-11-08:
Can confirm this persistently happening with some files. It may be related to files containing multiple lines like:

    python.code() # Some comment; With semicolons and 'apostrophes'
    python.code() # Some comment; With semicolons and 'apostrophes'