dhylands/rshell #149: Timeout Error with Raspberry Pi Pico - using version 0.0.29

RyanGaudion, 2021-03-01:
I'm using the most recent version released 2 days ago (0.0.29) on Windows 10 however I get timeouts whenever I try to copy files to my Pico. Here are the commands I have used:

`rshell`
`cd Desktop/`
`cp main.py /pyboard/main.py`

I have also tried these variations of the rshell command:

`rshell -a`
`rshell -p COM3 --buffer-size 512`
Both of which also timeout (timeout waiting for first EOF reception)

Saving the files through Thonny works as intended so the file itself shouldn't be an issue.

RyanGaudion, 2021-03-01:
Copy of Command Line:

```
C:\Users\user>rshell
Connecting to COM3 (buffer-size 128)...
Trying to connect to REPL  connected
Retrieving sysname ... rp2
Testing if sys.stdin.buffer exists ... Y
Retrieving root directories ... /main.py/
Setting time ... Mar 01, 2021 20:22:50
Evaluating board_name ... pyboard
Retrieving time epoch ... Jan 01, 1970
Welcome to rshell. Use the exit command to exit rshell.
C:\Users\user> cd Desktop/
C:\Users\user/Desktop> cp main.py /pyboard/main.py
Copying 'C:\Users\user/Desktop/main.py' to '/pyboard/main.py' ...
Traceback (most recent call last):
  File "C:\Users\user\AppData\Roaming\Python\Python37\Scripts\rshell-script.py", line 11, in <module>
    load_entry_point('rshell==0.0.29', 'console_scripts', 'rshell')()
  File "C:\Users\user\AppData\Roaming\Python\Python37\site-packages\rshell\command_line.py", line 4, in main
    rshell.main.main()
  File "C:\Users\user\AppData\Roaming\Python\Python37\site-packages\rshell\main.py", line 3084, in main
    real_main()
  File "C:\Users\user\AppData\Roaming\Python\Python37\site-packages\rshell\main.py", line 3067, in real_main
    shell.cmdloop(cmd_line)
  File "C:\Users\user\AppData\Roaming\Python\Python37\site-packages\rshell\main.py", line 1872, in cmdloop
    cmd.Cmd.cmdloop(self)
  File "c:\program files (x86)\microsoft visual studio\shared\python37_64\lib\cmd.py", line 138, in cmdloop
    stop = self.onecmd(line)
  File "C:\Users\user\AppData\Roaming\Python\Python37\site-packages\rshell\main.py", line 1902, in onecmd
    self.onecmd_exec("".join(group))
  File "C:\Users\user\AppData\Roaming\Python\Python37\site-packages\rshell\main.py", line 1913, in onecmd_exec
    return cmd.Cmd.onecmd(self, line)
  File "c:\program files (x86)\microsoft visual studio\shared\python37_64\lib\cmd.py", line 217, in onecmd
    return func(arg)
  File "C:\Users\user\AppData\Roaming\Python\Python37\site-packages\rshell\main.py", line 2329, in do_cp
    if not cp(src_filename, dst_filename):
  File "C:\Users\user\AppData\Roaming\Python\Python37\site-packages\rshell\main.py", line 685, in cp
    filesize, xfer_func=send_file_to_remote)
  File "C:\Users\user\AppData\Roaming\Python\Python37\site-packages\rshell\main.py", line 1616, in remote
    output, _ = self.pyb.follow(timeout=20)
  File "C:\Users\user\AppData\Roaming\Python\Python37\site-packages\rshell\pyboard.py", line 218, in follow
    raise PyboardError('timeout waiting for first EOF reception')
rshell.pyboard.PyboardError: timeout waiting for first EOF reception
```

davehylands, 2021-03-01:
I think I screwed up the 0.0.29 release. Can you try 0.0.30 instead?

Also, you shouldn't have to specify a buffer size. It should default to 128 (which is the default for the pico).

RyanGaudion, 2021-03-02:
Hi Dave, thanks for the quick reply - just tried with v0.0.30 and it throws the same error:

```
C:\Users\user>rshell --v
0.0.30

C:\Users\user>rshell
Connecting to COM3 (buffer-size 128)...
Trying to connect to REPL  connected
Retrieving sysname ... rp2
Testing if sys.stdin.buffer exists ... Y
Retrieving root directories ...
Setting time ... Mar 02, 2021 13:33:55
Evaluating board_name ... pyboard
Retrieving time epoch ... Jan 01, 1970
Welcome to rshell. Use the exit command to exit rshell.
C:\Users\user> cd Desktop/
C:\Users\user/Desktop> cp main.py /pyboard/main.py
Copying 'C:\Users\user/Desktop/main.py' to '/pyboard/main.py' ...
Traceback (most recent call last):
  File "C:\Users\user\AppData\Roaming\Python\Python37\Scripts\rshell-script.py", line 11, in <module>
    load_entry_point('rshell==0.0.30', 'console_scripts', 'rshell')()
  File "C:\Users\user\AppData\Roaming\Python\Python37\site-packages\rshell\command_line.py", line 4, in main
    rshell.main.main()
  File "C:\Users\user\AppData\Roaming\Python\Python37\site-packages\rshell\main.py", line 3084, in main
    real_main()
  File "C:\Users\user\AppData\Roaming\Python\Python37\site-packages\rshell\main.py", line 3067, in real_main
    shell.cmdloop(cmd_line)
  File "C:\Users\user\AppData\Roaming\Python\Python37\site-packages\rshell\main.py", line 1872, in cmdloop
    cmd.Cmd.cmdloop(self)
  File "c:\program files (x86)\microsoft visual studio\shared\python37_64\lib\cmd.py", line 138, in cmdloop
    stop = self.onecmd(line)
  File "C:\Users\user\AppData\Roaming\Python\Python37\site-packages\rshell\main.py", line 1902, in onecmd
    self.onecmd_exec("".join(group))
  File "C:\Users\user\AppData\Roaming\Python\Python37\site-packages\rshell\main.py", line 1913, in onecmd_exec
    return cmd.Cmd.onecmd(self, line)
  File "c:\program files (x86)\microsoft visual studio\shared\python37_64\lib\cmd.py", line 217, in onecmd
    return func(arg)
  File "C:\Users\user\AppData\Roaming\Python\Python37\site-packages\rshell\main.py", line 2329, in do_cp
    if not cp(src_filename, dst_filename):
  File "C:\Users\user\AppData\Roaming\Python\Python37\site-packages\rshell\main.py", line 685, in cp
    filesize, xfer_func=send_file_to_remote)
  File "C:\Users\user\AppData\Roaming\Python\Python37\site-packages\rshell\main.py", line 1616, in remote
    output, _ = self.pyb.follow(timeout=20)
  File "C:\Users\user\AppData\Roaming\Python\Python37\site-packages\rshell\pyboard.py", line 218, in follow
    raise PyboardError('timeout waiting for first EOF reception')
rshell.pyboard.PyboardError: timeout waiting for first EOF reception
```

davehylands, 2021-03-02:
So the fix that makes things work better for some people is this line here:
https://github.com/dhylands/rshell/blob/e2820f679aa06d40c35dbd5c626a0402047e28f8/rshell/main.py#L1107

It shows as a comment, but the rshell code "uncomments" any lines that start with `#rp2:` when the board is a Pico. If you run with `rshell -d` then you should see that line as uncommented when copying a file.

You could try increasing/decreasing that timeout to see if it helps you.

If you clone the github repository:
```
git clone https://github.com/dhylands/rshell.git
```
You can run rshell from the local tree by using:
```
cd rshell
./r.py ...use whatever rshell options you want here...
```

RyanGaudion, 2021-04-19:
@davehylands - thank you for your response and sorry for the long delay in getting back to you. Changing line 11088 to the below resolved the issue:
`#rp2: time.sleep_ms(50) `

RyanGaudion, 2021-04-19:
@davehylands - I've created a pull request (#154) for this - let me know if I have missed anything within the PR

davehylands, 2021-04-21:
@RyanGaudion PR looks fine. I just need to find some time to make a new release.

NitroBAY, 2021-11-25:
In my case increasing the buffer size saved me https://github.com/dhylands/rshell/pull/168

willbelr, 2021-12-16:
> In my case increasing the buffer size saved me #168

I got the same problem with the Pico. The delay did not help, but setting RPI_PICO_USB_BUFFER_SIZE to 256 did work.

MasterYuan418, 2023-01-19:
> 

Same as you, `PyboardError('timeout waiting for first EOF reception')`
on 0.0.31, while coping an 256KB file to Pico. Windows 10

reunano, 2023-08-07:
Im getting this error on a LEGO large hub : timed out or error in transfer to remote: b''
Will you be able to guide me on how to trouble shoot? Thank you!

willbelr, 2023-08-07:
> Im getting this error on a LEGO large hub : timed out or error in transfer to remote: b'' Will you be able to guide me on how to trouble shoot? Thank you!

Try using `rshell --buffer-size 256`

reunano, 2023-08-08:
Thank you for trying to help. This did not solve the issue I still get: /remote/remote.mpy' to '/pyboard/spike/remote.mpy' ...
timed out or error in transfer to remote: b''

ilium007, 2024-12-20:
Same here on shell 0.0.36 and Pico W

```
rshell --buffer-size 256 --port /dev/cu.usbmodem2301 -d
Debug = True
Port = /dev/cu.usbmodem2301
Baud = 115200
User = micro
Password = python
Wait = 0
List = 0
nocolor = 0
ascii = 0
Timing = 0
Quiet = 0
BUFFER_SIZE = 256
Cmd = []
Using buffer-size of 256
Connecting to /dev/cu.usbmodem2301 (buffer-size 256)...
Trying to connect to REPL  connected
Retrieving sysname ... ----- About to send 205 bytes of code to the pyboard -----
def sysname():
    try:
        import os
        return repr(os.uname().sysname)
    except:
        return repr('unknown')
output = sysname()
if output is None:
    print("None")
else:
    print(output)

-----
-----Response-----
b"'rp2'\r\n"
-----
rp2
Testing if ubinascii.unhexlify exists ... ----- About to send 286 bytes of code to the pyboard -----
def test_unhexlify():
    try:
        import ubinascii
    except:
        import binascii as ubinascii
    try:
        _ = ubinascii.unhexlify
        return True
    except:
        return False
output = test_unhexlify()
if output is None:
    print("None")
else:
    print(output)

-----
-----Response-----
b'True\r\n'
-----
Y
Retrieving root directories ... ----- About to send 150 bytes of code to the pyboard -----
def listdir(dirname):
    import os
    return os.listdir(dirname)
output = listdir('/')
if output is None:
    print("None")
else:
    print(output)

[... 147 lines cut ...]
                        bytes_read = sys.stdin.readinto(read_buf, read_size)
                    time.sleep_ms(20)
                    if bytes_read > 0:
                        write_buf[buf_index:bytes_read] = read_buf[0:bytes_read]
                        buf_index += bytes_read
                        buf_remaining -= bytes_read
                if False:
                    dst_file.write(write_buf[0:read_size])
                else:
                    dst_file.write(ubinascii.unhexlify(write_buf[0:read_size]))
                if hasattr(os, 'sync'):
                    os.sync()
                bytes_remaining -= read_size
        return True
    except:
        return False
output = recv_file_from_host(None, '/microdot.py', 966996)
if output is None:
    print("None")
else:
    print(output)

-----
timed out or error in transfer to remote: b'F'
```