dhylands/rshell #152: Line disappears when pressing backspace multiple times after a space

k4u5h1k, 2021-03-25:
## Do I have a solution?
**Yes**

## Details
Setting self.prompt to a single whitespace
 `self.prompt = ' '`   

instead of empty string 
`self.prompt = ''` 

in `line 1859` of `rshell/main.py`
fixes the issue completely on OSX

jamesquilty, 2022-07-22:
Thanks for this, that single-character addition does indeed fix the "disappearing prompt" problem that's been troubling me! I note that an artefact of this is that there are now two spaces `␣` after the prompt: `prompt>␣␣`. A quick hack that fixes this is to remove the trailing space from [line 1863 and adding it to line 1866 in the current HEAD revision on master](https://github.com/dhylands/rshell/blob/b87878c01aa02d6cd871ff4666983dee5584af16/rshell/main.py#L1861-L1871) appears to fix both problems:

```diff
     def set_prompt(self):
         if self.stdin == sys.stdin:
-            prompt = PROMPT_COLOR + cur_dir + END_COLOR + '> '
+            prompt = PROMPT_COLOR + cur_dir + END_COLOR + '>'
             if FAKE_INPUT_PROMPT:
                 print(prompt, end='')
-                self.prompt = ''
+                self.prompt = ' '
             else:
                 self.prompt = prompt
         else:
```

k4u5h1k, 2022-07-22:
Yep thats my PR :)