dhylands/rshell #157: Pico: file creation time is reported incorrectly

peterhinch, 2021-05-06:
I'm pretty sure this is a bug in rshell rather than in the firmware. In any event, it stops rsync from working with recently created files because they are assigned a datetime one hour ahead of local time.

This script works as expected, suggesting that the firmware is OK:
```python
import os, time

with open('/rats.txt', 'w') as f:
    f.write('rats\n')

s = os.stat('/rats.txt')
print(s)
print('file time', time.localtime(s[7]))
print('local time', time.localtime())
```
Which when pasted at the REPL produced this output:
```
5
(32768, 0, 0, 0, 0, 0, 5, 1620320616, 1620320616, 1620320616)
file time (2021, 5, 6, 17, 3, 36, 3, 126)
local time (2021, 5, 6, 17, 3, 36, 3, 126)
```
IOW the file was created at 17:03 which was correct. However in rshell I see
```bash
/mnt/qnap2/data/Projects/MicroPython/micropython-nano-gui> ls -l /pyboard/
     5 May  6 18:03 rats.txt
/mnt/qnap2/data/Projects/MicroPython/micropython-nano-gui> 
```
one hour ahead.

I am running Ubuntu 20.04. Local time here in the UK is UTC+1.

peterhinch, 2021-05-07:
This is not deterministic. Today rsync worked correctly until I power cycled the Pico. Weird.

njenwei, 2021-05-10:
I just encountered this issue too - someone correct me if I'm wrong, but it seems like micropython on the pico does not have timezones, and rshell sets the time to display correctly in gmtime and localtime. This means that the unix timestamp is offset by your timezone's offset.

Just as a work around, I edited this [line](https://github.com/dhylands/rshell/blob/e2820f679aa06d40c35dbd5c626a0402047e28f8/rshell/main.py#L1004) in the library to account for my system's offset `if stat_mtime(src_stat) > (stat_mtime(dst_stat) - -time.timezone):`.

peterhinch, 2021-05-11:
MicroPython running on a microcontroller does not have timezones. There are two pertinent observations.

- I have been a heavy user of **rsync** on STM, ESP32 and ESP8266 since its inception and have never encountered this issue. I am convinced it is Pico specific: it happens very frequently on that platform.
- As above, it is not deterministic.

ArmyAntSEC, 2021-06-27:
Same issue for me, but so far extremely consistent.
Unfortunately, @njenwei 's workaround did not seem to work. But then, python package building isn't my strength, so maybe I failed to install up updated version properly.

ArmyAntSEC, 2021-06-28:
Looking again at the code, I think the true issue is on line 1529: `self.time_offset = calendar.timegm(epoch_tuple)`
On my computer (Windows, and definately not on GMT) this return 0, meaning the program thinks there is no offset, which is clearly wrong. Since I saw the exact same issue on my Mac, I believe that the same issue should be happening there as well (But I don't have a debugger there, unfortunately)

ArmyAntSEC, 2021-06-28:
So I get this issue perfectly consistently, and it seems quite clear to me that it is because the Pico uses GMT time. On line 1653 the board is syned to the hosts localtime. But then unix timestamps are passed back that are assumed to be if UTC, but they are in fact not and that is what causes the issue.

For the pico specifically, I have checked that by simply syncing the board to GMT time, all of the logic becomes perfectly consistent and it all worked like a charm. unfortunately, I don't have a pyboard and I assume that the fact that the pyboard uses local time (That is what it says in the source comments) that this fix would completely break support for those boards.

I am still a bit unclear though what exactly the stat() function would return for a pyboard. Does it return the unix timestamp, but with the timezone offset added? This confuses me, and stops me from developing a patch that would actually work for both clock types.

ArmyAntSEC, 2021-06-28:
So if anyone else is affected, replacing line 1653 with `now = time.gmtime(time.time())` seems to work really well, both in practice, and when looking at the way the code works.

peterhinch, 2021-06-29:
It's unclear if the problem is in **rshell** or the Pico firmware. Even with today's daily build, **rshell** is reporting a file creation date one hour ahead of _local_ time (i.e. UTC +2). I remain puzzled by my original observation re **os.stat**.

It seems to me that all ports should behave similarly. I would argue that they should use local time. For a user in a timezone far from UTC, anything else is going to be quite confusing. If that point of view is accepted, then the original code in **rshell** is correct. Bear in mind that it works with ESP32, ESP8266 and all Pyboards; probably other ports too.

In summary I don't think we've yet got to the bottom of the real cause of this issue.

moefear85, 2021-09-06:
There seems to be a smore sinister bug writhling in the guts of rshell somewhere. When I update a file "loop.py", rshell fails to update it on the esp8266. It furthermore updates a file "hostname.py" even though I never touched that file. Then I discover when I do "rsync . /pyboard" several times, it keeps updating the same files, even though they should have already been updated the first time, and it is not even clear if it actually physically updated those files. I discovered by chance later that it didn't touch boot.py even though I updated its local copy on my PC, and it reported it has having been updated.

This isn't usable. It's leading to so many error messages and headaches.

peterhinch, 2021-09-07:
I have used rshell heavily for several years, including rsync, and never seen the behaviour you describe. I strongly suspect there is another cause. Is the file hostname.py something to do with webrepl? Could this be updating it?