dhylands/rshell #169: [Licensing] Consider dropping autoconnection-based "pyudev" support

leycec, 2021-12-11:
While packaging `rshell` for Gentoo Linux, <sup>:smiling_face_with_three_hearts:</sup> I reluctantly noticed that [`pyudev` is LGPL-licensed](https://pyudev.readthedocs.io/en/latest/licencing.html). <sup>:face_with_head_bandage:</sup>

Since `rshell` imports from `pyudev`, the infectious transitivity of the LGPL means that `rshell` *must* also be (L)GPL-licensed. But `rshell` is currently MIT-licensed. This contradiction can be resolved in one of three ways (in increasing order of soul-rending painfulness):

* **(Recommended)** Drop autoconnection-based `pyudev` support entirely. This is what you want to do.
* Relicense `rshell` under the LGPL. This is harder than you think, because you'd legally need the explicit consent of *everyone* who's ever contributed even a single line of code to `rshell` to do so. This is usually *not* what you want to do.
* Do nothing, cross fingers, and pretend everything will be okay. Maybe denial *is* a river in Egypt, after all? But it's not. So, this is the hardest and darkest road you can ever walk. Stallman's Free Software Foundation (FSF) religiously enforces (L)GPL violations. You *will* eventually be forced to either:
  * Walk one of the two alternate roads outlined above *or*...
  * **Delete this entire repository.**

I know what I'd do – but I also know I'm thankfully not @dhylands. Gods. I hate delivering issues like this.

leycec, 2021-12-11:
**Nevermind!** My face is now shamefully plastered with egg, because the entire point of the LGPL is that it enables linking (importation, in this case) *without* viral infection. The coast is clear, folks. `</facepalm>`

closed 2021-12-11 by leycec

davehylands, 2021-12-17:
Thanks for bringing up the issue and the resolution. I think its cool that rshell is being included in the Gentoo Linux distribution.