dhylands/rshell #171: pyreadline no longer works under Windows 10 with Python 3.10

dhylands, 2021-12-13:
pyreadline no longer works under Windows 10 with Python 3.10 but works with python 3.9

See discussion in the forum: https://forum.micropython.org/viewtopic.php?f=15&t=11606

leycec, 2022-01-01:
**Yuppers.** IPython dropped `pyreadline` seven years ago – and it's been increasingly dead ever since. `rshell` now needs to follow IPython's by refactoring the codebase to use the popular [Python Prompt Toolkit](https://github.com/prompt-toolkit/python-prompt-toolkit) instead.

Fortunately, Python Prompt Toolkit is *substantially* superior to `pyreadline`. Unfortunately, Python Prompt Toolkit isn't simply a drop-in replacement for `pyreadline`; it's really a whole new kettle of CLI fish. Fortunately, `rshell` only references `pyreadline` **exactly 6 times** in its `rshell.main` module.

So, this refactoring *should* be trivial – in theory. Someone just needs to get down to brass tacks, roll up their sleeves, and make this happen in 2022 before Python 3.10 goes mainstream. `</gulp>`

moefear85, 2022-01-04:
A temporary fix would be to edit the following file (replace "PYTHON" with the path to your python installation):

PYTHON/Lib/site-packages/pyreadline/py3k_compat.py

then modify line 8 from:
```
return isinstance(x, collections.Callable)
```

so it becomes:
```
return isinstance(x, collections.abc.Callable)
```

save it, and thats it. when you launch rshell the next time, problem solved. The reason of the problem is that the python library moved the location of "Callable" from the collections module into the "abc" submodule. Thus the outdated pyreadline can't find it.

leycec, 2022-01-05:
[`pyreadline` is no longer maintained. Its last commit was seven years ago.](https://github.com/pyreadline/pyreadline) Since upstream is dead, upstream won't merge anything – which means the only means of meaningfully applying this change would be for `rshell` itself to:

1. Vendor (i.e., bundle, embed) `pyreadline` inside itself.
2. Apply the trivial patch you've helpfully supplied, @moefear85.
3. **Maintain this vendored copy of `pyreadline` in perpetuity.**

In the short term, that's probably the sanest solution. In the long term, `rshell` absolutely should *not* be vendoring unmaintained and possibly insecure third-party packages inside itself.

[Python Prompt Toolkit](https://github.com/prompt-toolkit/python-prompt-toolkit) is still the way.

gamefunc, 2023-11-29:
fix pr: https://github.com/dhylands/rshell/pull/223
win 10 python3.11 build: https://github.com/gamefunc/rshell/releases/tag/v0.0.32