dhylands/rshell #173: PICO XIAO - timed out or error in transfer to remote: b'F'

pacmac, 2021-12-24:
As below, I seem to be unable to copy any files to the device.

Also, even setting the --buffer-size=128 gets ignored and the 512 USB default is used.

Files are created with zero bytes, I also tried the --ascii option but it made no difference.

Any suggestions ??

```
> rshell --buffer-size=128
Connecting to /dev/cu.usbmodem0000000000001 (buffer-size 512)...
Trying to connect to REPL  connected
Retrieving sysname ... unknown
Testing if sys.stdin.buffer exists ... N
Retrieving root directories ... /servo.py/ /bq2060.py/ /notes.txt/ /bqfunc.py/ /bqconfig.py/
Setting time ... Dec 24, 2021 14:44:58
Evaluating board_name ... pyboard
Retrieving time epoch ... Jan 01, 2000
Welcome to rshell. Use Control-D (or the exit command) to exit rshell.
> ls /pyboard -all
     0 Jan  1 2000  bq2060.py
     0 Jan  1 2000  bqconfig.py
     0 Jan  1 2000  bqfunc.py
     0 Jan  1 2000  notes.txt
     0 Jan  1 2000  servo.py
> cp flash/bq2060.py /pyboard
Copying '/Users/pac/docker/upydev/work/projects/aibo-xiao/flash/bq2060.py' to '/pyboard/bq2060.py' ...
timed out or error in transfer to remote: b'F'
>
```

dmccue128, 2022-01-02:
I have the same problem with my esp32s2 (feathers2).  
Running rshell on my Windows 11 Home PC, when I try "cp foo.py /pyboard/", it creates the directory entry for foo.py on the esp32, but it is zero bytes and the cp command fails with "timed out or error in transfer to remote: b''"

sadilekivan, 2022-01-23:
Same problem on Linux 5.15 with Raspberry Pi Pico and Qt-Py RP2040 (micropython 1.18), creates an empty file. Can't think of a solution, but for the meantime I started using [mpfshell](https://github.com/wendlers/mpfshell).

dhylands, 2022-01-24:
There was an issue with the --buffer-size that caused it to be ignored on the Rpi. I've just fixed that and it's available in version 0.0.31

That version also sets the default buffersize to be 32 for the RPi (the definition of an RPi in this context is a device with a USB VID:PID of 2e8a:0005)

jose1711, 2022-08-25:
So this can be closed?

dmccue128, 2022-08-25:
Yes, it can be closed, thank you.