dhylands/rshell #177: rsync -m: say goodbye to your build host filesystem data

cederom, 2022-01-13:
Hello world :-)

* New to MicroPython. 
* Using `rshell` 0.0.31.
* Saw `rshell help rsync`:
```
rshell help rsync

No MicroPython boards connected - use the connect command to add one

usage: rsync [-m|--mirror] [-n|--dry-run] [-q|--quiet] SRC_DIR DEST_DIR

Synchronizes a destination directory tree with a source directory tree.

positional arguments:
  SRC_DIR        Source directory
  DEST_DIR       Destination directory

optional arguments:
  -h, --help     show this help message and exit
  -a, --all      Don't ignore files starting with .
  -m, --mirror   causes files in the destination which don't exist in the source to be removed. Without --mirror only file copies occur. No deletions will
                 take place.
  -n, --dry-run  shows what would be done without actually performing any file copies. Implies --verbose.
  -q, --quiet    Doesn't show what has been done.
```
* Did use `SRC_DIR=.` and `DEST_DIR=/`.
```
rshell -p /dev/cuaU0 rsync -m . /
```
* Said goodbye to a local user files :-(

**This makes `rshell` extremely dangerous application when user provides invalid argument it will corrupt local machine without check or warning**. 
* Situation should not be possible to take place.
* Please consider preventing in any way local filesystem destruction.
* Not sure if `rsync` should operate on a local filesystem as target at all (maybe simple board connection and board name check + prompt against `DEST_DIR` on `rsync` start would do the job).

cederom, 2022-01-16:
This happened again, script did not provide variable to `rshell rsync -m . /$VARIABLE`, and `rshell` started deleting my all files outside project. This program and its syntax is a joke. Removing and will never come back. This tool is a threat to user files and system. Beware.

dhylands, 2022-01-17:
Yep - you said you wanted / which is the root of your your local computer, not located on the pyboard. i.e. very similar to `rm -rf /` will also remove all of the files on your machine.

I'll see what I can do about figuring out what "local" paths might be considered dangerous (/ and $HOME are obvious candidates). rsync should also probably complain if the source and destination are both local.

dhylands, 2022-01-17:
If you're not sure what directory you're using, you should use `ls` and verify that it's the directory that you think it is.

cederom, 2022-01-18:
Hey @dhylands :-) 

Maybe that sounds funny to create an application that will result in `rm -rf /` but I really lost lots of my files then lots of time to retrieve them to loose them again, first when learning `rshell`, second when using it in a script just to make sure that I get the board name right so then because of connectivity issues did not return the board name and it launched `rshell rsync -m . /` resulting in `rm -rf /`.

This syntax is a disaster I tell you do not use anything (i.e. `/`) that may even confuse local filesystem and the board filesystem.

For instance `adafruit-ampy` points only to local files while command tells you what to do with the board and the local file. This seems the safest approach. Another example is `mpremote` that clearly distinguishes between local filesystem (`/`) and board filesystem (`:`).

I hope it is not necessary for you one day to try out some new application out there that will remove all your files to understand what I mean.

Take care :-)

dhylands, 2022-01-19:
I agree. In this case rsync is trying to model the behaviour of the regular rsync command available in linux, and if you ran `rsync -m . /` if would also proceed to remove all of your files.

rshell distinguishes between local and remote resources by using the path and I can't really change that now without breaking lots of people's scripts. I think it should disallow / as a destination for some of the commands (like rm, cp, or rsync).

rshell's rsync command also has a --dry-run option (which neither copies nor removes files - just prints what it would do) which you probably should have considered using while you were testing your script.

TomBurg, 2023-03-29:
Hi Dave, Have you tried using rshell on any ESP32-S3 boards from Adafruit?

dhylands, 2023-04-09:
> Hi Dave, Have you tried using rshell on any ESP32-S3 boards from Adafruit?

I can't say that I have. I don't often use ESP32's

TG9541, 2024-07-18:
I tried `rshell` on a generic ESP32C3. It doesn't work as described, and in fact it sometimes doesn't work at all - it's apparently random if files on the target board appear, it's unlikely that I can do anything but `cat` a file contents if it appears. The likeness of folders doesn't work, etc. 

Since the theory of operation or the normal behavior are not clear to a naive user, and since it's obviously not sufficiently tested, it has the potential of doing very bad things to the local file system, I now consider `rshell` a dangerous program that shouldn't be advertised for usage by anyone but by `rshell` developers.

dhylands, 2024-08-12:
I've been using rshell on the ESP32-S3 with no issues.

/pyboard is the default location of the ESP32 file system (as documented on the rshell page). So you should be doing things like:
`cp somefile.py /pyboard`

VisualGadget, 2024-08-24:
I'm searched for this issue after rsync -m some /   Blame me for docs skipping.
I propose to block rsync run when SRC or DEST is not belongs to a python device folder.