dhylands/rshell #180: rshell not working

pem120, 2022-03-17:
Traceback (most recent call last):
  File "C:\Users\IshantKuchibhatla\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.10_qbz5n2kfra8p0\LocalCache\local-packages\Python310\Scripts\rshell-script.py", line 33, in <module>
    sys.exit(load_entry_point('rshell==0.0.29', 'console_scripts', 'rshell')())
  File "C:\Users\IshantKuchibhatla\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.10_qbz5n2kfra8p0\LocalCache\local-packages\Python310\Scripts\rshell-script.py", line 25, in importlib_load_entry_point
    return next(matches).load()
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.10_3.10.752.0_x64__qbz5n2kfra8p0\lib\importlib\metadata\__init__.py", line 162, in load
    module = import_module(match.group('module'))
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.10_3.10.752.0_x64__qbz5n2kfra8p0\lib\importlib\__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1050, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1027, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1006, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 688, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 883, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "C:\Users\IshantKuchibhatla\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.10_qbz5n2kfra8p0\LocalCache\local-packages\Python310\site-packages\rshell\command_line.py", line 1, in <module>
    import rshell.main
  File "C:\Users\IshantKuchibhatla\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.10_qbz5n2kfra8p0\LocalCache\local-packages\Python310\site-packages\rshell\main.py", line 80, in <module>
    import readline
  File "C:\Users\IshantKuchibhatla\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.10_qbz5n2kfra8p0\LocalCache\local-packages\Python310\site-packages\readline.py", line 34, in <module>
    rl = Readline()
  File "C:\Users\IshantKuchibhatla\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.10_qbz5n2kfra8p0\LocalCache\local-packages\Python310\site-packages\pyreadline\rlmain.py", line 422, in __init__
    BaseReadline.__init__(self)
  File "C:\Users\IshantKuchibhatla\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.10_qbz5n2kfra8p0\LocalCache\local-packages\Python310\site-packages\pyreadline\rlmain.py", line 62, in __init__
    mode.init_editing_mode(None)
  File "C:\Users\IshantKuchibhatla\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.10_qbz5n2kfra8p0\LocalCache\local-packages\Python310\site-packages\pyreadline\modes\emacs.py", line 633, in init_editing_mode
    self._bind_key('space',       self.self_insert)
  File "C:\Users\IshantKuchibhatla\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.10_qbz5n2kfra8p0\LocalCache\local-packages\Python310\site-packages\pyreadline\modes\basemode.py", line 162, in _bind_key
    if not callable(func):
  File "C:\Users\IshantKuchibhatla\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.10_qbz5n2kfra8p0\LocalCache\local-packages\Python310\site-packages\pyreadline\py3k_compat.py", line 8, in callable
    return isinstance(x, collections.Callable)
AttributeError: module 'collections' has no attribute 'Callable'

anegin13, 2022-03-25:
Same error here. Any clues?
> rshell -h

Traceback (most recent call last):
  File "C:\portable\Python\Python310\Scripts\rshell-script.py", line 33, in <module>
    sys.exit(load_entry_point('rshell==0.0.31', 'console_scripts', 'rshell')())
  File "C:\portable\Python\Python310\Scripts\rshell-script.py", line 25, in importlib_load_entry_point
    return next(matches).load()
  File "C:\portable\Python\Python310\lib\importlib\metadata\__init__.py", line 162, in load
    module = import_module(match.group('module'))
  File "C:\portable\Python\Python310\lib\importlib\__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1050, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1027, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1006, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 688, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 883, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "C:\portable\Python\Python310\lib\site-packages\rshell\command_line.py", line 1, in <module>
    import rshell.main
  File "C:\portable\Python\Python310\lib\site-packages\rshell\main.py", line 80, in <module>
    import readline
  File "C:\portable\Python\Python310\lib\site-packages\readline.py", line 34, in <module>
    rl = Readline()
  File "C:\portable\Python\Python310\lib\site-packages\pyreadline\rlmain.py", line 422, in __init__
    BaseReadline.__init__(self)
  File "C:\portable\Python\Python310\lib\site-packages\pyreadline\rlmain.py", line 62, in __init__
    mode.init_editing_mode(None)
  File "C:\portable\Python\Python310\lib\site-packages\pyreadline\modes\emacs.py", line 633, in init_editing_mode
    self._bind_key('space',       self.self_insert)
  File "C:\portable\Python\Python310\lib\site-packages\pyreadline\modes\basemode.py", line 162, in _bind_key
    if not callable(func):
  File "C:\portable\Python\Python310\lib\site-packages\pyreadline\py3k_compat.py", line 8, in callable
    return isinstance(x, collections.Callable)
AttributeError: module 'collections' has no attribute 'Callable'

pem120, 2022-03-26:
no.
Traceback (most recent call last):
  File "C:\Users\IshantKuchibhatla\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.10_qbz5n2kfra8p0\LocalCache\local-packages\Python310\Scripts\rshell-script.py", line 33, in <module>
    sys.exit(load_entry_point('rshell==0.0.29', 'console_scripts', 'rshell')())
  File "C:\Users\IshantKuchibhatla\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.10_qbz5n2kfra8p0\LocalCache\local-packages\Python310\Scripts\rshell-script.py", line 25, in importlib_load_entry_point
    return next(matches).load()
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.10_3.10.1008.0_x64__qbz5n2kfra8p0\lib\importlib\metadata\__init__.py", line 171, in load
    module = import_module(match.group('module'))
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.10_3.10.1008.0_x64__qbz5n2kfra8p0\lib\importlib\__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1050, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1027, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1006, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 688, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 883, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "C:\Users\IshantKuchibhatla\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.10_qbz5n2kfra8p0\LocalCache\local-packages\Python310\site-packages\rshell\command_line.py", line 1, in <module>
    import rshell.main
  File "C:\Users\IshantKuchibhatla\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.10_qbz5n2kfra8p0\LocalCache\local-packages\Python310\site-packages\rshell\main.py", line 80, in <module>
    import readline
  File "C:\Users\IshantKuchibhatla\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.10_qbz5n2kfra8p0\LocalCache\local-packages\Python310\site-packages\readline.py", line 34, in <module>
    rl = Readline()
  File "C:\Users\IshantKuchibhatla\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.10_qbz5n2kfra8p0\LocalCache\local-packages\Python310\site-packages\pyreadline\rlmain.py", line 422, in __init__
    BaseReadline.__init__(self)
  File "C:\Users\IshantKuchibhatla\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.10_qbz5n2kfra8p0\LocalCache\local-packages\Python310\site-packages\pyreadline\rlmain.py", line 62, in __init__
    mode.init_editing_mode(None)
  File "C:\Users\IshantKuchibhatla\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.10_qbz5n2kfra8p0\LocalCache\local-packages\Python310\site-packages\pyreadline\modes\emacs.py", line 633, in init_editing_mode
    self._bind_key('space',       self.self_insert)
  File "C:\Users\IshantKuchibhatla\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.10_qbz5n2kfra8p0\LocalCache\local-packages\Python310\site-packages\pyreadline\modes\basemode.py", line 162, in _bind_key
    if not callable(func):
  File "C:\Users\IshantKuchibhatla\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.10_qbz5n2kfra8p0\LocalCache\local-packages\Python310\site-packages\pyreadline\py3k_compat.py", line 8, in callable
    return isinstance(x, collections.Callable)
AttributeError: module 'collections' has no attribute 'Callable'

pem120, 2022-03-26:
Traceback (most recent call last):
  File "C:\Users\IshantKuchibhatla\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.10_qbz5n2kfra8p0\LocalCache\local-packages\Python310\Scripts\rshell-script.py", line 33, in <module>
    sys.exit(load_entry_point('rshell==0.0.29', 'console_scripts', 'rshell')())
  File "C:\Users\IshantKuchibhatla\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.10_qbz5n2kfra8p0\LocalCache\local-packages\Python310\Scripts\rshell-script.py", line 25, in importlib_load_entry_point
    return next(matches).load()
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.10_3.10.1008.0_x64__qbz5n2kfra8p0\lib\importlib\metadata\__init__.py", line 171, in load
    module = import_module(match.group('module'))
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.10_3.10.1008.0_x64__qbz5n2kfra8p0\lib\importlib\__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1050, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1027, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1006, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 688, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 883, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "C:\Users\IshantKuchibhatla\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.10_qbz5n2kfra8p0\LocalCache\local-packages\Python310\site-packages\rshell\command_line.py", line 1, in <module>
    import rshell.main
  File "C:\Users\IshantKuchibhatla\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.10_qbz5n2kfra8p0\LocalCache\local-packages\Python310\site-packages\rshell\main.py", line 80, in <module>
    import readline
  File "C:\Users\IshantKuchibhatla\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.10_qbz5n2kfra8p0\LocalCache\local-packages\Python310\site-packages\readline.py", line 34, in <module>
    rl = Readline()
  File "C:\Users\IshantKuchibhatla\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.10_qbz5n2kfra8p0\LocalCache\local-packages\Python310\site-packages\pyreadline\rlmain.py", line 422, in __init__
    BaseReadline.__init__(self)
  File "C:\Users\IshantKuchibhatla\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.10_qbz5n2kfra8p0\LocalCache\local-packages\Python310\site-packages\pyreadline\rlmain.py", line 62, in __init__
    mode.init_editing_mode(None)
  File "C:\Users\IshantKuchibhatla\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.10_qbz5n2kfra8p0\LocalCache\local-packages\Python310\site-packages\pyreadline\modes\emacs.py", line 633, in init_editing_mode
    self._bind_key('space',       self.self_insert)
  File "C:\Users\IshantKuchibhatla\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.10_qbz5n2kfra8p0\LocalCache\local-packages\Python310\site-packages\pyreadline\modes\basemode.py", line 162, in _bind_key
    if not callable(func):
  File "C:\Users\IshantKuchibhatla\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.10_qbz5n2kfra8p0\LocalCache\local-packages\Python310\site-packages\pyreadline\py3k_compat.py", line 8, in callable
    return isinstance(x, collections.Callable)
AttributeError: module 'collections' has no attribute 'Callable'

no

dhylands, 2022-04-09:
I think that this is a problem with Windows a readline.

Does this workaround work: https://github.com/dhylands/rshell/pull/38#issuecomment-397832096

nznobody, 2022-06-03:
Possibly look a using an updated fork of pyreadline: git+https://github.com/osrf/pyreadline
As done here: https://github.com/ros2/ci/pull/519
Otherwise switch to https://github.com/pyreadline3/pyreadline3

anegin13, 2022-06-04:
> Possibly look a using an updated fork of pyreadline: git+https://github.com/osrf/pyreadline As done here: [ros2/ci#519](https://github.com/ros2/ci/pull/519) Otherwise switch to https://github.com/pyreadline3/pyreadline3

Installing pyreadline3 resolve the issue. Thank you.

Techmokid, 2023-07-21:
I've discovered the solution. "Callables" in python 3.10 or above has been moved from "collections.Callables" to "collections.abc.Callables"

The following code in the designated file "site-packages/pyreadline/py3k_compat.py" solved the issue for me, although newer versions of pyreadline3 may have already fixed this issue:

from __future__ import print_function, unicode_literals, absolute_import
import sys

if sys.version_info[0] >= 3:
    import collections
    PY3 = True
    def callable(x):
        if (sys.version_info[1] >= 10):
            return isinstance(x, collections.abc.Callable)
        else:
            return isinstance(x, collections.Callable)
    
    def execfile(fname, glob, loc=None):
        loc = loc if (loc is not None) else glob
        with open(fname) as fil:
            txt = fil.read()
        exec(compile(txt, fname, 'exec'), glob, loc)

    unicode = str
    bytes = bytes
    from io import StringIO
else:
    PY3 = False
    callable = callable
    execfile = execfile
    bytes = str
    unicode = unicode
    
    from StringIO import StringIO

aminhusni, 2024-06-02:
When I install `pip install pyreadline3`, I get this error:
```
Traceback (most recent call last):
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.10_3.10.3056.0_x64__qbz5n2kfra8p0\lib\runpy.py", line 196, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.10_3.10.3056.0_x64__qbz5n2kfra8p0\lib\runpy.py", line 86, in _run_code
    exec(code, run_globals)
  File "C:\Users\alice\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.10_qbz5n2kfra8p0\LocalCache\local-packages\Python310\Scripts\rshell.exe\__main__.py", line 4, in <module>
  File "C:\Users\alice\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.10_qbz5n2kfra8p0\LocalCache\local-packages\Python310\site-packages\rshell\command_line.py", line 1, in <module>
    import rshell.main
  File "C:\Users\alice\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.10_qbz5n2kfra8p0\LocalCache\local-packages\Python310\site-packages\rshell\main.py", line 80, in <module>
    import readline
ModuleNotFoundError: No module named 'readline'
```

When I install `pip install pyreadline`, I get this error:
```
Traceback (most recent call last):
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.10_3.10.3056.0_x64__qbz5n2kfra8p0\lib\runpy.py", line 196, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.10_3.10.3056.0_x64__qbz5n2kfra8p0\lib\runpy.py", line 86, in _run_code
    exec(code, run_globals)
  File "C:\Users\alice\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.10_qbz5n2kfra8p0\LocalCache\local-packages\Python310\Scripts\rshell.exe\__main__.py", line 4, in <module>
  File "C:\Users\alice\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.10_qbz5n2kfra8p0\LocalCache\local-packages\Python310\site-packages\rshell\command_line.py", line 1, in <module>
    import rshell.main
  File "C:\Users\alice\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.10_qbz5n2kfra8p0\LocalCache\local-packages\Python310\site-packages\rshell\main.py", line 80, in <module>
    import readline
  File "C:\Users\alice\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.10_qbz5n2kfra8p0\LocalCache\local-packages\Python310\site-packages\readline.py", line 34, in <module>
    rl = Readline()
  File "C:\Users\alice\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.10_qbz5n2kfra8p0\LocalCache\local-packages\Python310\site-packages\pyreadline\rlmain.py", line 422, in __init__
    BaseReadline.__init__(self)
  File "C:\Users\alice\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.10_qbz5n2kfra8p0\LocalCache\local-packages\Python310\site-packages\pyreadline\rlmain.py", line 62, in __init__
    mode.init_editing_mode(None)
  File "C:\Users\alice\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.10_qbz5n2kfra8p0\LocalCache\local-packages\Python310\site-packages\pyreadline\modes\emacs.py", line 633, in init_editing_mode
    self._bind_key('space',       self.self_insert)
  File "C:\Users\alice\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.10_qbz5n2kfra8p0\LocalCache\local-packages\Python310\site-packages\pyreadline\modes\basemode.py", line 162, in _bind_key
    if not callable(func):
  File "C:\Users\alice\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.10_qbz5n2kfra8p0\LocalCache\local-packages\Python310\site-packages\pyreadline\py3k_compat.py", line 8, in callable
    return isinstance(x, collections.Callable)
AttributeError: module 'collections' has no attribute 'Callable'
```

Currently there is no way to run this with the pyreadline3 workaround.

pmansouri-davey, 2024-08-16:
@aminhusni 

From: https://stackoverflow.com/a/74037892
> The issue is caused by the outdated package pyreadline present as a package importable by the Python installation. As this provides the readline module for Windows version of Python, the default site module will be able to import it and trigger the error. This outdated and no longer maintained package tries to import collections.abc.Callable from the deprecated location collections.Callable, [it will fail under Python 3.10](https://github.com/pyreadline/pyreadline/issues/73) as that deprecated import has been [fully removed](https://stackoverflow.com/questions/53978542/how-to-use-collections-abc-from-both-python-3-8-and-python-2-7).

In your case, the reason you are seeing the error `ModuleNotFoundError: No module named 'readline'` is most likely due to having install pyreadline3 when pyreadline was already installed.
Try uninstalling both and then reinstall 3:
`pip uninstall pyreadline pyreadline3`
`pip install pyreadline3`