dhylands/rshell #188: add -I optional argument to rsync to ignore date

vvladic, 2022-05-18:
It would be great if one could just mirror and overwrite the entire host folder using "rsync -m -I ./ /pyboard"

dhylands, 2022-05-18:
If you use `cp -r` it will recursively copy a directory and doesn't look at timestamps.

vvladic, 2022-05-18:
Normally, I would just use rsync without -I, but the thing is file timestamp on the Pico is showing 2 hours in future...

```
C:\Workspace\Micropython\pico> ls -la .
   293 May 18 19:45 blink.py
   255 May 18 19:46 main.py
   127 May 18 19:45 name_tools.py
C:\Workspace\Micropython\pico> ls -la /pyboard
   293 May 18 21:46 blink.py
   245 May 18 21:46 main.py
   127 May 18 21:46 name_tools.py
```

But ESP32 file timestamps seem to be the same as on my host machine.
```
C:\Workspace\Micropython\esp32> ls -la .
   289 May 18 20:14 blink.py
   137 May 18 20:14 boot.py
   245 May 18 20:14 main.py
   127 May 18 20:14 name_tools.py
C:\Workspace\Micropython\esp32> ls -la /pyboard
   289 May 18 20:12 blink.py
   137 May 18 20:12 boot.py
   245 May 18 20:12 main.py
   127 May 18 20:12 name_tools.py
```

cp -r says
```
C:\Workspace\Micropython\pico> cp -r ./*.py /pyboard
File 'C:\Workspace\Micropython\pico/C:\Workspace\Micropython\pico/blink.py' doesn't exist
```
for some reason...

vvladic, 2022-05-19:
On Ubuntu, the workaround with `cp -r` works fine. 
I will be using that until the problem with rp2 timestamps is fixed, after which I will just use `rsync`.
Thank you for your help.

closed 2022-05-19 by vvladic

dhylands, 2022-05-25:
> cp -r says
> 
> ```
> C:\Workspace\Micropython\pico> cp -r ./*.py /pyboard
> File 'C:\Workspace\Micropython\pico/C:\Workspace\Micropython\pico/blink.py' doesn't exist
> ```
> 
> for some reason...

That's a useful clue. rshell assumes that the file seperators will always be '/' and it looks like under windows the \ didn't get converted to / in the current working directory.

I wasn't aware of issues with timestamps being in the future on the pico, so I'll look into that as well.

rshell normallys sets the time on the microcontroller to match the time on the host so that when files are created on the microcontroller then they'll match up with timestamps from the host. But it's possible that the RPi Picto is doing something funky here.