dhylands/rshell #192: Unable to get exit code on cp/rsync

zzzbatmand, 2022-07-26:
I want to see if a cp/rsync fails, but when I try to use $? after the command has failed ([sys.exit(2)](https://github.com/dhylands/rshell/blob/master/rshell/main.py#L1137)) then I just get a 0.

OS: Windows WSL

```
me@PC:/mnt/c/Users/me/Documents/TMP/fw# rshell -p /dev/ttyS7 cp -r ./Pycom/* /flash
Using buffer-size of 32
Connecting to /dev/ttyS7 (buffer-size 32)...
Trying to connect to REPL  connected
Retrieving sysname ... GPy
Testing if ubinascii.unhexlify exists ... Y
Retrieving root directories ... /flash/
Setting time ... Jul 26, 2022 10:27:05
Evaluating board_name ... pyboard
Retrieving time epoch ... Jan 01, 1970
timed out or error in transfer to remote: b''
me@PC:/mnt/c/Users/me/Documents/TMP/fw# echo $?
0
```

**Sidenote:**
The way I tested this, was by disconnecting the serial wires doing upload.
Sometimes this caused the program to just hit an infinite loop.

neopostmodern, 2024-11-15:
Same on Ubuntu – this is a major bummer for me because I use this in scripts with `set -o errexit` and wrongly assumed the whole thing was successful when `rshell cp` had actually failed somewhere further up in the script.

dhylands, 2024-12-04:
Thanks for bring this up. This isn't a path I had considered. I'll see what I can do to fix this.

dhylands, 2024-12-04:
I think I've fixed it so that if that if you do `rshell some-command` and `some-command` encounters an error then the exit code will be non-zero (in version 0.0.36)

neopostmodern, 2024-12-08:
Thank you! I'll be working with this in the upcoming week and will report back!