dhylands/rshell #198: Unable to connect to Pico on OpenBSD

yuuwe-n, 2022-09-01:
I am trying to get the raspberry pi pico working on OpenBSD. I have gotten the firmware onto the pico and got minicom to work; However rshell is unable to connect.

```
Welcome to minicom 2.8

OPTIONS: I18n 
Compiled on Apr  8 2022, 08:49:03.
Port /dev/ttyU0, 16:18:42
```

```
rshell --buffer-size=512 -p /dev/ttyU0 
Using buffer-size of 512
Connecting to /dev/ttyU0 (23.217.138.110) ...
No response from 23.217.138.110
Welcome to rshell. Use Control-D (or the exit command) to exit rshell.

No MicroPython boards connected - use the connect command to add one

```

```
rshell -l                                                                                 
Serial Device: /dev/cuaU0                                                                          
Serial Device: /dev/cuaU1                                                                          
Serial Device: /dev/cuaU2                        
Serial Device: /dev/cuaU3                                                                          
Serial Device: /dev/cua00                                                                          
Serial Device: /dev/cua01                                                                          
Serial Device: /dev/cua02                                                                          
Serial Device: /dev/cua03                        
Serial Device: /dev/cua04                                                                          
Serial Device: /dev/cua05                        
Serial Device: /dev/cua06                                                                          
Serial Device: /dev/cua07                        
Serial Device: /dev/cua08                                                                          
Serial Device: /dev/cua09                      
Serial Device: /dev/cua0a                                                                          
Serial Device: /dev/cua0b                    
Serial Device: /dev/cuac0
Serial Device: /dev/cuac1
Serial Device: /dev/cuac2
Serial Device: /dev/cuac3
Serial Device: /dev/cuac4
Serial Device: /dev/cuac5
Serial Device: /dev/cuac6
Serial Device: /dev/cuac7
```
```
rshell -V  
0.0.31
```

I think its because of the port, but I am not sure why they minicom and rshell would be using different ports. Something similar happened when i was trying to get the arduino mega to work, so I tried connecting to /dev/cua* ports; however it would still give the same error

```
rshell --buffer-size=512 -p /dev/cuaU0 
Using buffer-size of 512
Connecting to /dev/cuaU0 (23.217.138.110) ...
No response from 23.217.138.110
Welcome to rshell. Use Control-D (or the exit command) to exit rshell.

No MicroPython boards connected - use the connect command to add one
```

Any thoughts on what to try next?

yuuwe-n, 2022-09-01:
I got `adafruit-ampy` to work using the ttyU0 port 

```
ampy --port /dev/ttyU0 get blink.py
```

dhylands, 2022-09-07:
rshell is trying /dev/cuaU0 as a hostname and getting an IP address back, which is confusing it (rshell supports connecting over telnet to some micropython boards). hostnames don't have slashes in them so I think I'll need to have skip the ip stuff if there is a slash in the name passed in.