dhylands/rshell #202: On Windows (and Linux?) rshell is significantly slower when `-p` is specified

arbego, 2022-11-08:
In PowerShell:

Without `-p` option: **0.76elapsed**

    time rshell --quiet 'exit'

    0.00user 0.00system 0:00.76elapsed 0%CPU (0avgtext+0avgdata 5140maxresident)k
    0inputs+0outputs (1338major+0minor)pagefaults 0swaps

With `-p` option: **1.77elapsed**

    time rshell --quiet -p COM6 'exit'

    0.00user 0.00system 0:01.77elapsed 0%CPU (0avgtext+0avgdata 5128maxresident)k
    0inputs+0outputs (1336major+0minor)pagefaults 0swaps

heeplr, 2023-01-11:
can confirm this on linux:

```
$ time rshell --debug --port /dev/ttyACM0 cp -r ./src/* "/pyboard"
Debug = True
Port = /dev/ttyACM0
Baud = 115200
User = micro
Password = python
Wait = 0
List = 0
nocolor = 0
ascii = 0
Timing = 0
Quiet = 0
BUFFER_SIZE = 512
Cmd = [cp, -r, ./src/main.py, /pyboard]
Using buffer-size of 32
Connecting to /dev/ttyACM0 (buffer-size 32)...
Trying to connect to REPL  connected
Retrieving sysname ... ----- About to send 205 bytes of code to the pyboard -----
def sysname():
    try:
        import os
        return repr(os.uname().sysname)
    except:
        return repr('unknown')
output = sysname()
if output is None:
    print("None")
else:
    print(output)

-----
-----Response-----
b"'rp2'\r\n"
-----
rp2
Testing if ubinascii.unhexlify exists ... ----- About to send 224 bytes of code to the pyboard -----
def test_unhexlify():
    import ubinascii
    try:
        _ = ubinascii.unhexlify
        return True
    except:
        return False
output = test_unhexlify()
if output is None:
    print("None")
else:
    print(output)

-----
-----Response-----
b'True\r\n'
-----
Y
Retrieving root directories ... ----- About to send 150 bytes of code to the pyboard -----
def listdir(dirname):
    import os
    return os.listdir(dirname)
output = listdir('/')
if output is None:
    print("None")
else:
    print(output)

-----
-----Response-----
b"['config.json', 'main.py']\r\n"
[... 435 lines cut ...]
                        write_buf[buf_index:bytes_read] = read_buf[0:bytes_read]
                        buf_index += bytes_read
                        buf_remaining -= bytes_read
                if True:
                    dst_file.write(write_buf[0:read_size])
                else:
                    dst_file.write(ubinascii.unhexlify(write_buf[0:read_size]))
                if hasattr(os, 'sync'):
                    os.sync()
                bytes_remaining -= read_size
        return True
    except:
        return False
output = recv_file_from_host(None, '/main.py', 1018)
if output is None:
    print("None")
else:
    print(output)

-----
-----Response-----
b'True\r\n'
-----
rshell --debug cp -r ./src/* "/pyboard"  0,21s user 0,03s system 30% cpu 0,775 total
```

github0null, 2024-08-11:
Hello,

I have also found this problem,

This line will not work on Windows when '-p COMx' is specified

https://github.com/dhylands/rshell/blob/56f75cf846b2fd89f9a89c08a9dabd8015c30931/rshell/main.py#L1422

I have post a PR to fix it. 

https://github.com/dhylands/rshell/pull/232