dhylands/rshell #221: creation time is incorrect for files uploaded to the board

dkonyshev, 2023-09-17:
Hello,

I tried to use `rsync` to upload updated files in rshell with my RPi Pico board and it didn't work for me. The problem is that my timezone is +6 to UTC and all files uploaded to the board get creation time in future. It seems that the problem occurs because rshell uses the current local time to set the current time on the board. As Micropython doesn't have any timezone information, the time is interpreted as UTC time.

For now I fixed this like follows but, perhaps, there's a better way?

```
$ git diff
diff --git a/rshell/main.py b/rshell/main.py
index 7a810d9..80526fb 100755
--- a/rshell/main.py
+++ b/rshell/main.py
@@ -1657,7 +1657,7 @@ class Device(object):

     def sync_time(self):
         """Sets the time on the pyboard to match the time on the host."""
-        now = time.localtime(time.time())
+        now = time.gmtime(time.time())
         self.remote(set_time, (now.tm_year, now.tm_mon, now.tm_mday, now.tm_wday + 1,
                                now.tm_hour, now.tm_min, now.tm_sec, 0))
         return now
```

laulin, 2023-09-21:
Hello dkonyshev,

Same problem and you patch work. Thank you !

A side effect is now the log line "Setting time ..." do not display the right time : it shift at the opposite time. 

Example before : 
Setting time ... 4pm [ the right time]
File timestamp : 6pm

Example now :
Setting time ... 2pm 
File timestamp : 4pm [ the right time]

bablokb, 2024-06-08:
I think this is a misunderstanding on how timestamps are stored in the filesystem: they are always stored as UTC and the programs that display the timestamps have to do the translation (or better: interpretation) of the timestamps. Just run the `stat` command on any local file and you will see what I mean: you get a timestamp that is in UTC but your computer will show you the correct time.

So rshell is correct. If you  use the "ls -l" command from rshell, you will see correct times. If you check the timestamp with some other program (e.g. via a mount), you won't see the correct time.